ppsp-team/HyPyP #17: Better "pip install HyPyP"

deep-introspection, 2020-05-23:
Include all necessary requirements following "pip install HyPyP"

YannBeauxis, 2020-05-27:
Poetry could really helps you to automatically includes required packages on the build ;)
https://python-poetry.org/docs/
I've already add a poetry config file on the branch master.

YannBeauxis, 2020-05-27:
Did you work on versions scope compatibility for all packages ?

In requirements.txt :

- some packages have no version specifications

- some are pinned a a unique version

- some use ">=" and should have a protection to major version change with "<" such as:
`matplotlib>=3.0.2, <4.0.0` rather than `matplotlib>=3.0.2`

Ideally, you should even explore compatibility with lower or higher existing versions to have the broader requirement scope. That because hypyp could be used in a project with other tools that have their own requirements for shared libraries and could create incompatibility issues.

closed 2020-06-24 by deep-introspection