ppsp-team/HyPyP #18: `dir(hypyp)` get an empty list of function.

AmirDJV, 2020-05-26:
When I run `dir(hypyp)` or `inspect` I get an empty list of function. 

[image]

[image]

YannBeauxis, 2020-05-27:
What functions do you expect to get ?
To attach some attributes or function to a module you should use `__init__.py` to import and use `__all__` attribute like that :
```
from .analyses import PSD

__all__ = ["PSD"]
```

AmirDJV, 2020-05-27:
According the [`dir` docs](https://docs.python.org/3/library/functions.html#dir), I'm supposed to get a list of attributes. 
For example, If I'll run `dir(mne)` I will get: 

[image]

deep-introspection, 2020-06-24:
@YannBeauxis Do you know how we can solve this?

YannBeauxis, 2020-06-24:
Yes in the root `__init__.py` file, put some : 

```
from hypyp import viz, prep

__all__ = ["viz", "prep"]
```

I let you put what you want to get with dir(hypyp)

deep-introspection, 2020-06-25:
Thanks @YannBeauxis! @AmirDJV does it work now with the new Github version?

closed 2020-09-22 by deep-introspection