ppsp-team/HyPyP #22: Error for importing 3D headmodel

aayrolles, 2020-06-04:
if it reports an error regarding obj missing when plotting 3D inter-brain connectivity, try to move “data/Basehead.obj” to "C:\ProgramData\Miniconda3\Lib\site-packages\"

aayrolles, 2020-06-26:
Here the error @ufangyang gets, we check, the headmodel data are in the good repository and the link work for the epoch file 
[image]

deep-introspection, 2020-06-26:
This is coming from ```os.pardir``` i.e. ".." in the path. The problem is that the ```__file__``` position and the ```data``` folder are not relatively place similarly as in the Github repository structure.

For the epochs, how does ```os.path.join(...)``` look like?

aayrolles, 2020-06-26:
Here are the both path : 

mesh = meshio.read(os.path.join(os.path.dirname(__file__),
           os.pardir,'data',"Basehead.obj"))

epo1 = mne.read_epochs(
    os.path.join(os.path.dirname(__file__), os.pardir, "data", "participant1-epo.fif"),
    preload=True,
)

aayrolles, 2020-06-26:
Maybe it come from the ' ' instead of " "?

deep-introspection, 2020-06-26:
'' vs "" shouldn't have an impact in Python.

Maybe this is the read function of meshio that does not like ```..``` in the path. Try to check how MNE read_epochs is handling the string of characters. Maybe they have a function transforming ```/A/B/../C/``` into ```/A/C/```

YannBeauxis, 2020-06-26:
You should use [pathlib](https://docs.python.org/3/library/pathlib.html) ilbrary with Path.resolve() method.

In general I prefer pathlib.Path rather than os.path ;)

aayrolles, 2020-06-27:
Thanks Yann ! I change it for ` mesh = meshio.read(Path('../data/Basehead.obj').resolve())`
It should work better now

closed 2020-06-27 by aayrolles

ufangYang, 2020-06-29:
The problem has been solved for me only if I specify the path for `Basehead.obj` in the `getting_started` script.

Below is the piece of code in `getting_started` that has generated the error as abovementioned (i.e., the screenshot). 

`   mesh = meshio.read(os.path.join(os.path.dirname(__file__),
           os.pardir,'data',"Basehead.obj"))`

I changed this code to

`    mesh=meshio.read(
    Path('../data/Basehead.obj').resolve())`

Then instead of using `vertices, faces = viz.get_3d_heads()` I used `vertices, faces = get_3d_heads()`

Changing these lines, I am now able to plot in 2D and 3D.