ppsp-team/HyPyP #25: error in fit_params for ca_fit function

ufangYang, 2020-06-16:
it seems there is a type error in ICA_fit() for the argument 'fit_params', even when I drop off the `extended=TRUE` or change to 'FALSE'

ihgumilar, 2020-06-17:
There are two points here :
1. There is no explanation for `fit_params` when we press shift + tab
2. When I dropped the `fit_params` the ICA works. It seems that the argument of fit_params is not included in the function ?

[image]

aayrolles, 2020-06-17:
I don't know why but you were true, the problem came from HyPyP version. We uninstall HyPyP and reinstall it with the `pip install HyPyP`. 
I think it could have been fix even without installing poetry.

closed 2020-06-18 by deep-introspection