ppsp-team/HyPyP #27: Fix mpl3d dependency

YannBeauxis, 2020-06-17:
`mpl3d` ( https://github.com/rougier/matplotlib-3d ) is required for `viz` module but is currently available only with git and does not exist in PyPi. This lib is quite eager and its maintainability need to be secure for a stable version of HyPyP.

- [ ] Check commitment and plans for the lib of the mpl3d developers

- [ ] deploy the mpl3d on PyPi

aayrolles, 2020-07-22:
Hi, We have modified the visualization to no longer be dependent on mapl3d. For the moment in `master` the two possibilities of visualization still exist. And we can use either headmodel from mpl3D or the new and more simple topographic map. But we also can decide to delete completely mpl3D for futher version.

deep-introspection, 2020-07-22:
@aayrolles Maybe the best would be to check with @Rougier what are his plans for the mpt3d library. If it's put on pypi, then we can keep it.

rougier, 2020-07-23:
That's the mid-term plan but not before September I would say. In the meantime ang given the small size of mpl3d, you might consider to include the source in your package. Can you open an issue on mpl3D for requesting the PyPi (else I might forget)?

closed 2020-09-22 by deep-introspection