ppsp-team/HyPyP #29: Intra-brain visualization

deep-introspection, 2020-06-24:
Add possibility to also plot intra-individual connectivity in the visualizations

ighui, 2020-07-09:
Hi, Guillaume. I have two questions when I use getting_started.py. 
The first is when I pip install the master and git+https://github.com/rougier/matplotlib-3d ,there is error in my anaconda terminal 'ERROR: Could not install packages due to an EnvironmentError: HTTPSConnectionPool(host='api.github.com', port=443): Max retries exceeded with url: /repos/autoreject/autoreject/zipball/master (Caused by ProtocolError('Connection aborted.', ConnectionResetError' and  while executing command git clone -q https://github.com/rougier/matplotlib-3d 'C:\Users\NATURE~1\AppData\Local\Temp\pip-req-build-4gaoflqq' ERROR: Cannot find command 'git' - do you have 'git' installed and in your PATH?
The second is error 'file does not start with a file id tag'.

deep-introspection, 2020-07-09:
Hi @ighui! Those errors does not seem related to HyPyP, and even less to this issue of intra-brain visualization. Can you please start a dedicated thread and better describe your environment, os, etc. since those error look related to your specific installation and internet connection especially.

closed 2020-09-22 by deep-introspection