ppsp-team/HyPyP #32: viz.get_3d_heads() error

phoebsc, 2020-07-01:
I got this error while running getting_started.
It seems that the ../data folder isn't included in the pip installed package, so Basehead.obj is missing.

---------------------------------------------------------------------------
ReadError                                 Traceback (most recent call last)
<ipython-input-34-e961320b9a12> in <module>
      1 fig, ax = plt.subplots(1, 1)
      2 ax.axis("off")
----> 3 vertices, faces = viz.get_3d_heads()
      4 camera = Camera("ortho", theta=90, phi=180, scale=1)
      5 mesh = Mesh(ax, camera.transform @ glm.yrotate(90), vertices, faces,

c:\users\phoebe chen\appdata\local\programs\python\python37\lib\site-packages\hypyp\viz.py in get_3d_heads()
    490     # Extract vertices and faces for the first head
    491     mesh = meshio.read(os.path.join(os.path.dirname(__file__),
--> 492            os.pardir,'data',"Basehead.obj"))
    493     zoom = 0.064
    494     interval = 0.32

c:\users\phoebe chen\appdata\local\programs\python\python37\lib\site-packages\meshio\_helpers.py in read(filename, file_format)
     55         path = pathlib.Path(filename)
     56         if not path.exists():
---> 57             raise ReadError("File {} not found.".format(filename))
     58 
     59         if not file_format:

**ReadError: File c:\users\phoebe chen\appdata\local\programs\python\python37\lib\site-packages\hypyp\..\data\Basehead.obj not found.**

YannBeauxis, 2020-07-02:
I can include the data folder to the pip package but it turns hypyp from a 22kB to a 17MB package.
It also may be possible to get the data from github with urls.

phoebsc, 2020-07-02:
Is it possible to only include Basehead.obj? I don't think the sample files are needed.

phoebsc, 2020-07-02:
Great! Thanks!

closed 2020-07-02 by phoebsc