ppsp-team/HyPyP #40: Partial directed coherence

sofievettori, 2020-11-23:
In the paper is mentioned that directed measures of causality or information transfer are implemented. But it doesn't seem to be in the source code of the current version. Is there a newer version available with it implemented? Or do you know when it will be available? 
thank you!

deep-introspection, 2020-11-25:
For now, the PDC is not supported but we hope to implement it for the upcoming Beta release. If you are willing to help, do not hesitate to propose a python implementation that we can insert in the relevant function: i.e. https://github.com/ppsp-team/HyPyP/blob/master/hypyp/analyses.py#L391 (ping @phoebsc!)

deep-introspection, 2020-11-25:
Maybe the MNE team has already found out a good way to get PDC: https://gist.github.com/agramfort/9875439

AmirDJV, 2020-11-25:
Going over MNE's [issue](https://github.com/mne-tools/mne-python/issues/1187), it seems like at some point they wanted to compare it to [SCoT](https://scot-dev.github.io/scot-doc/index.html)'s [PDC function](https://scot-dev.github.io/scot-doc/api/scot/scot.html#scot.Connectivity.PDC). From what I understand they did some testings for both functions. It might be interesting to see if the results are the same and under different conditions. If they are, to go with the easier implantation and benchmark.  What to do you think?

deep-introspection, 2021-04-06:
@phoebsc, we just had a nice pull request from @ghazalehran! Can you please double-check her proposal?

closed 2021-11-10 by deep-introspection