ppsp-team/HyPyP #44: Error in 3D visualization: Basehead.obj not found

DiGyt, 2021-01-12:
When trying to create a 3D visualization, e.g. with `hypyp.viz.viz_3D_inter`, I get an Error reading `ReadError: File /my_current_working_directory/data/Basehead.obj not found.`, which prevents me from using the 3D plotting functions.

I think the reason for this is that [this line]( https://github.com/ppsp-team/HyPyP/blob/aeda650b975cecc9b01bb3b12877091c3519a260/hypyp/viz.py#L595) in hypyp.viz looks at the relative path of my working directory instead of the relative path from hypyp's base directory.

Problem occurred when loading Hypyp via Jupyter Notebook.

deep-introspection, 2021-04-06:
Thanks, we are looking into this!

deep-introspection, 2021-04-06:
@DiGyt Can you please try again with the Beta.2 version?

closed 2021-04-06 by deep-introspection