ppsp-team/HyPyP #48: Incorrect 2D Topomap Visualization

delara-aryan, 2021-03-29:
Hi,

When I run the viz_2D_topomap_inter function from viz.py after processing my data using all the steps in the getting_started notebook, instead of a result as shown in the tutorial, I get what looks like 2 overlapping brains (the electrodes are shown on top of each other):
[image]
I am loading data from edf files, and manually adding channel location data to each electrode's Epoch info object as shown:
```
locations = pd.read_table('129Channels.xyz', skiprows=1, delim_whitespace=True, names=['X', 'Y', 'Z', 'Name'])
for subject in [epochs1.info["chs"], epochsd.info["chs"]]:
    for electrode in subject:
        name = electrode["ch_name"]
        row = locations.loc[locations.Name == name]
        ref_row = locations.loc[locations.Name == "E129"]
        x = float(row.X)
        y = float(row.Y)
        z = float(row.Z)
        cx = float(ref_row.X)
        cy = float(ref_row.Y)
        cz = float(ref_row.Z)
        electrode["loc"] = np.array([x, y, z, cx, cy, cz, 0, 0, 0, 0, 0, 0])
```

Could there be an issue in the format I am loading data or adding channel location information that is causing this output for the visualization?

delara-aryan, 2021-03-31:
To change the visualization output, I tried adjusting the traX, traY, and traZ values for each time the transform() function is called on the electrode locations in viz.py by multiplying each value by 100 as shown:
`loc1 = transform(loc1, traX=-0.17*100, traY=0, traZ=0.08*100, rotY=(-np.pi/12), rotZ=(-np.pi/2))`

This resulted in a slightly better output for viz_2D_topomap_inter in which the electrode locations are no longer overlapping; however, the links between them still appear to be incorrect:
[image]

Also, even with this change, I am still getting bad results for viz_3D_inter:
[image]

deep-introspection, 2021-04-06:
Dear @delara-aryan, can you provide a sample of 1-epoch file with your sensors position so I can make a test?

closed 2021-11-04 by deep-introspection