ppsp-team/HyPyP #55: PLV in negative value

ihgumilar, 2021-05-27:
Hi Hypyp team,

As we see the visualization that it shows a blue line, which indicates a negative correlation (red line is for positive correlation).
However, for Phase Locking Value (PLV) analysis, I believe that the range of the score is between 0 and 1. If so, how can the visualization give a blue line, which indicates a negative value (which is not in the range of PLV analysis) ? Please advise.

Thanks
IG

ihgumilar, 2021-06-09:
Hi,

Thanks for the reply. 

I am a bit confused about the C (Cohen's D value) here. How can we interpret the C value in this case ? isn't it aimed to see 
 how much a difference between two participants ? If so, I don't see any substraction between two participants' means instead it was just subtracted from the mean of alpha_low values. Since this looks at connectivity between two participants, I don't think we focus on a difference of means between two participants, please correct me if I am wrong.

Also, regarding **Comparing Inter-brain connectivity values to random signal** in line 44 in notebook, the result says

> Using a threshold of 18.512821
> stat_fun(H1): min=inf max=inf
> Running initial clustering
> **Found 1 clusters**
> Permuting 4999 times...

However, when I plotted with the following code, I did not show any line that indicated a significant value of the above cluster. Could you give me some help ?
```
viz.viz_2D_topomap_inter(epo1, epo2, statscondCluster.F_obs_plot, 
                         threshold='auto', steps=10, lab=True)
```
[image]

Thanks

deep-introspection, 2021-11-30:
@ihgumilar It should be solved now. Let us know if you get another problem.

closed 2021-11-30 by deep-introspection