ppsp-team/HyPyP #63: Incorrect 2D and 3D visualization

MissDan1995, 2021-10-18:
Hi Hypyp team,

When I run the viz_2D_topomap_inter and 3D _inter functions following all the steps in the getting_started notebook, I get what looks like 2 overlapping brains (the electrodes are shown on top of each other):
[image]
[image]
```
 viz.viz_2D_topomap_inter(epochs1, epochs2, C, threshold='auto', steps=10, lab=True)
 viz.viz_3D_inter(epochs1, epochs2, C, threshold='auto', steps=10, lab=False)
```
I don't know how to solve this issue, could you help me?@aayrolles

MissDan1995, 2021-10-18:
Here are sample data
https://github.com/MissDan1995/hypyp_test_data.git

katied-neuro, 2021-10-21:
This seems like maybe an issue with setting your montage and channel locations. Are you able to provide your script and .locs file with your channel locations?

MissDan1995, 2021-10-22:
> This seems like maybe an issue with setting your montage and channel locations. Are you able to provide your script and .locs file with your channel locations?
I use this file( standard-10-5-cap385.elp) to implement channel location  in EEGLAB.
Here are file links https://github.com/MissDan1995/hypyp_test_data.git

katied-neuro, 2021-11-05:
Hello @MissDan1995, 

I know that we are hoping to put out a tutorial on EEGlab data soon. 

I took a look at your code and it does not look like your are setting your montage.

eeglab_montage="../standard-10-5-cap385.elp"   #set your montage to the file containing your channel location
montage=mne.channels.read_custom_montage(eeglab_montage) 

#Now apply this montage to your data
epochs1.set_montage(montage)
epochs2.set_montage(montage)

#I would also plot the sensors to double check if the montage is what you expect.
epochs1.plot_sensors()

Then, you should be able to continue analyzing your data. 

But- I know you also had issues with converting to a .fif format earlier, so I'm worried that unless that issue was fixed, what I just suggested will not fix your problem.

Let me know if that works!

deep-introspection, 2021-11-30:
@MissDan1995 It should be solved now. Let us know if you get another problem.

closed 2021-11-30 by deep-introspection