ppsp-team/HyPyP #64: Read XDF file containing dyadic EEG data in HyPyP

FranckPrts, 2021-11-08:


FranckPrts, 2021-11-15:
How can I read dyadic EEG data embedded in [XDF](https://github.com/sccn/xdf#extensible-data-format-xdf) formatted files?

In my case, the XDF files to read were generated by [LSL](https://labstreaminglayer.readthedocs.io/info/intro.html) and contain six streams (e.g., 4 videos streams and 2 EEG streams). 

First, my objective is to read the XDF, detect the EEG streams embedded within, and, while respecting their metadata (name, sampling rate, etc.), convert them into MNE `Raw` objects. Following this, my second objective would be to align these two EEG streams using the time-stamps provided by the XDF and enable connectivity analysis.

FranckPrts, 2022-04-07:
## Update 
Beta version of the import pipeline for XDF can be found [here](https://github.com/FranckPrts/HyPyP/tree/xdf-imports).

See
- The [tutorial](https://github.com/FranckPrts/HyPyP/blob/xdf-imports/tutorial/test_xdf_import.ipynb)
- The new class at the bottom of [utils.py](https://github.com/FranckPrts/HyPyP/blob/xdf-imports/hypyp/utils.py)

patricefortin, 2025-04-17:
Hi @FranckPrts ,

I'm new member of the HyPyP dev team and a part of the recent effort to bring bugfixes and new features to the project.

I have been looking at your branch and the code for XDF_IMPORT. It's a nice feature and a clearly a welcomed addition to the project. 

Given the new coding convention we are putting in place, would you agree if I start a new branch on the main project, heavily inspired by your work (giving you credits), and from there I would work on your "second objective" ?

> _Following this, my second objective would be to align these two EEG streams using the time-stamps provided by the XDF #and enable connectivity analysis._

FranckPrts, 2025-04-21:
Hi Patrice,
Yes, absolutely; I totally agree and I'm really happy to know this piece will see the light of day! Thanks for picking this up and giving it the attention it deserves! 🙌
Feel free to reach out if you want to discuss anything else or need anything else. I'll stay tuned to the repo to see these new coding convention too!

jonasmago, 2025-04-23:
Hey Patrice, great to see you here!
I can just second Franck here, super happy for you to go ahead with that plan. 
I think we ended up being almost done with it and it was just a matter of integrating it well in the final code. I'd also be super happy to jump on a live coding call for an hour or two to wrap this up together. 
Let me know if that would be helpful at all or if you have all you need to just get it done. 
Jonas

patricefortin, 2025-04-23:
It would be useful if you could commit to the project the XDF files that are used in the tutorial. 

I can otherwise generate xdf with synthetic data (what I did for running the tutorials locally), but it would be best to have real ones.