ppsp-team/HyPyP #71: Cross-Wavelet Transform

deep-introspection, 2021-11-16:
[Python implementation](https://pycwt.readthedocs.io/en/latest/)

deep-introspection, 2021-11-16:
Later on, consider cross-frequency.

claytonjschneider, 2022-06-08:
@deep-introspection is there any version of wavelet transform coherence analysis currently in HyPyP? I assumed there was, and on looking at compute_sync, I can't tell but assume the 'coh' connectivity measure refers to cross-spectral coherence? 

I am planning to do my analysis using wavelet transform coherence analysis and I am happy to contribute to this toolbox however possible, if the feature does not exist yet.

katied-neuro, 2022-06-14:
Hi- we are currently in the process of working on some wavelet based analysis tools based on the cross wavelet transform described in Torrence and Compo's 1998 paper (https://paos.colorado.edu/research/wavelets/). Currently, this works for EEG, but I am still working on some bugs for fNIRS. Later this week, I am planning to upload what I have so far to my branch of HyPyP as a rough draft and provide a link to that here so that you can see what I have so far and if it suits your needs.

claytonjschneider, 2022-06-15:
Awesome, thank you for the update @katied-neuro . If you happen to think of it, please @ me here when you push that to your branch! I can do my best to help debug / show you how it works for our data.

katied-neuro, 2022-07-06:
Hi @claytonjschneider,

Here is the wavelet draft. I need to add error messages to warn people about duplicate/identical datasets and using too small a time frame for the analyses. I also need to program the plots so that the COI is indicated (i.e. the area not fit for use). 

Let me know what you think. 
https://github.com/katied-neuro/HyPyP/blob/master/tutorial/fnirs-wavelets-2022-07-06kd.ipynb 

Thanks!

claytonjschneider, 2022-07-06:
Hi @katied-neuro !! Thank you so much! I will take a look now :D appreciate you sharing this with me and all that you are doing at HyPyP

claytonjschneider, 2022-07-07:
@katied-neuro do I need to be using a specific branch to access this code for now? If it's the branch under your name, I don't know that I can access it. I got a 404 when trying to look.

claytonjschneider, 2022-07-07:
Nice, looks like that works. So I'm going to give this a shot now by cloning the repo, rather than installing via pip, and checkout to your branch, and import the library from sys path before import. Hopefully that works?

katied-neuro, 2022-07-07:
Are you talking about my repository or HyPyP in general?I have mne 0.24.0. and looking at HyPyP, it is currently set-up according to the requirements.txt for mne 0.24.0.

claytonjschneider, 2022-07-07:
OK I'm back to mne 0.24.0, hopefully that works as I've ran into problems with the previous versions of MNE loading NIRx data before. 

Also - how are you adding your local HyPyP branch to your python modules? I tried adding it to the path with sys.path.append, then import hypyp, and got this error:  `DistributionNotFound: The 'hypyp' distribution was not found and is required by the application`. 

Then I tried running `pip install .` from inside the cloned repository, but got an error that the wheel building failed. 

I can't seem to find a setup.py if one exists - so how do I import your branch?

claytonjschneider, 2022-07-07:
Aweome! I messaged you on twitter