ppsp-team/HyPyP #75: What input does analyses.compute_freq_bands expect?

MLTOL, 2021-12-07:
Hi,

Congratulations on your HyPyp pipeline. I am trying to implement your HyPyP pipeline for our own data. Unfortunately, I have an issue with the analysis pipeline. 

We use EEG signals obtained by the MUSE S. Data is stored in XDF files, which I converted to mne objects. Since we use the MUSE S the montage is standard 1020, but only the TP9, TP10, AF7 and AF8 are included. I included our events in our data and created an epoch object based on these events. I was able to run the preprocess pipeline correctly. 

Please, could you tell me what data format is required for the methods functions in the analyses pipeline? We were able to apply the autoreject. Then, following your manuscript, I thought I should use your merge and split function to do some additional data preparation. However, when I try to run the code I get a data shape error:

[image]

[image]

Alternatively, I tried to input the epochs outputted by the autoreject directly into the compute_freq_bands function. But then I got the following error:

[image]
[image]

Apparently, the function does not expect the mne object, but an array. I know the compute_freq_bands function expects a shape of (2, n_epochs, n_channels, n_times). What data from your pipeline does this function expect? How do I get from the output of the autoreject function to the expected data format?

I hope you can help. Thank you for your time.

katied-neuro, 2021-12-15:
Hello, 

I think the issue is that you need to feed in an np array to compute_freq_bands that contains your two sets of epochs, similar to the data_inter framework shown in the tutorial in box 21 onwards (under connectivity). 

https://github.com/ppsp-team/HyPyP/blob/master/tutorial/getting_started.ipynb 

Please let us know if this works :)

deep-introspection, 2022-02-24:
We will close that issue for now. Let us know if the bug is still there.

closed 2022-02-24 by deep-introspection