ppsp-team/HyPyP #119: Line 172 `values -= np.diag(np.diag(values))` should be delet

MissDan1995, 2022-05-13:
https://github.com/ppsp-team/HyPyP/blob/00114e786ece59138b94a395366ce4afde5a60a0/tutorial/getting_started.py#L172
Line 172 `values -= np.diag(np.diag(values))`
I think you should delete line172, otherwise the synchrony data for the same channel in the interbrain part of the matrix will be deleted.

deep-introspection, 2022-05-16:
No, it's not an error here. We actually removed intentionally the diagonal because all the values are equal to 1.0 and thus are not informative and interfere with the colormap span in the visualization.

closed 2022-05-16 by deep-introspection

MissDan1995, 2022-05-17:
I agree with your points. But for **inter** brain connectivity the diagonal values are not euqal to 1.0.    Only the diagonal values of **intra** brain connectivity are equal to 1.0.