ppsp-team/HyPyP #138: Integrate automatic labeling of ICA component into preprocessing

ihgumilar, 2022-07-28:
Hi ,

Since it was just released this [new feature](https://mne.discourse.group/t/mne-icalabel-v0-2-is-released-on-pypi/5185) to label automatically ICA component (whether it is brain, eye blink, heart beat, and others), that would be great if we could integrate that feature into` ICA_choice_comp()` function which is under preprocessing. People can choose whether they want to choose manually or automatically. However, if you have a suitable place to put it, feel free to do so.

It would help us to save the time if we have hundreds of files instead of looking at hundreds of ICA components manually. Hope that is useful.

Thanks

ihgumilar, 2022-08-04:
This has been integrated [here](https://github.com/ppsp-team/HyPyP/pull/141#issue-1328037980)

deep-introspection, 2022-08-04:
Nice, thanks!

I would just separate the ICA_fit from the correction. My guess would be to merge the functions [ICA_choice_comp](https://github.com/ppsp-team/HyPyP/blob/master/hypyp/prep.py#L42) and [ICA_apply](https://github.com/ppsp-team/HyPyP/blob/master/hypyp/prep.py#L91) into a new ICA_correct function, with the option to do _manual_ vs _automatic_ labeling of the components.

What do you think @ihgumilar ?

ihgumilar, 2022-08-04:
That's awesome !

I have been thinking about it as well, but I thought you may want to separate it in the beginning.
I totally agree with that, that would make things much simpler. We just need to explain it a bit more in the documentation later on.

Please let me know if there is anything that I can help further.

deep-introspection, 2022-08-05:
So here is my proposal: https://github.com/ppsp-team/HyPyP/pull/142/commits/b21d8c9e74b7249060ef49e4044be6fdc3b790f3#diff-ce8e2bd69734b3a4016f90f228f16b6f8fa9f0647b4c0b917c328ee761101b3c

ihgumilar, 2022-08-06:
Is it only this part that I screenshot ?
Did you upload it in master or another branch ? I am trying to update on my side but nothing happens.

[image]

ihgumilar, 2022-08-06:
Yeah, that is fine. I tested.
But, one option that we need to provide, I think, we should give an option whether user wants to visualize the epoch data **before VS after** cleaning ICA components (using automatic labeling function). It would be much more useful and informative for the user, particularly for those who have a visual style of learning.

deep-introspection, 2022-08-08:
Like this? https://github.com/ppsp-team/HyPyP/commit/a8e4023ba6a2d324f237d4e96ffda209a533911b

ihgumilar, 2022-08-08:
Lovely @deep-introspection ...Much better code for showing the graph.
Thanks for updating the tutorial as well.

Great..!

deep-introspection, 2022-08-09:
Ok, great! I close this issue then. Thanks @ihgumilar

closed 2022-08-09 by deep-introspection