ppsp-team/HyPyP #160: compute_single_freq returns an array with a wrong shape

m-agat, 2023-03-24:
Hi Team,

When I run compute_single_freq function from hypyp analyses it returns a 6-dimensional array instead of a 5d array. I run it with my own data and the data you provided in the getting_started notebook and the array I get is of shape (2, n_epochs, n_channels, 3 n_frequencies, n_times). Do you have an idea why I get that extra 3 in the shape of the array? 

This is the code I run in the getting_started notebook:

# with ICA
data_inter = np.array([preproc_S1, preproc_S2])

complex_sig = analyses.compute_single_freq(data_inter, sampling_rate, freq_bands["Alpha-Low"])
print(complex_sig.shape)

Output:
(2, 34, 31, 3, 4, 250)

deep-introspection, 2023-05-12:
Greetings,
Well spotted! MNE output actually a dimension for the multitaper methods of n_tapers size.
I will change the documentation of `compute_single_freq` accordingly.

deep-introspection, 2023-05-12:
Solved in https://github.com/ppsp-team/HyPyP/commit/724a8b0323231691fa6607a29b1417ed1016ae31

closed 2023-05-12 by deep-introspection