ppsp-team/HyPyP #164: Questions about the usage of stats.statscluster

parkinsonlyn, 2023-05-15:
Hi HyPyP team, thank you for your great efforts. Having read the introductory paper and the tutorial, I still have some questions regarding the usage of stats.statscluster to achieve cluster-level permutation and hoping to receive some advice.

As I have already calculated the inter-brain connectivity between all dyads using Ccorr, and have sliced and stored the inter-brain part of the matrix. Now I would like to see whether there are differences between two groups in inter-brain connectivity. The other arguments seem no problem except for ch_con_freq, which needs to be calculated for the metaconnectivity (as I did not calculate PSD
or CSD prior).  Thus according to the tutorial, the calculation is as follow:
```
# targeted frequency as Theta
freqs_mean = np.arange(4, 8)

# merge two example data to cal. indicies
subjall = utils.merge(subj0, subj1)

# cal. indicies of inter-brain connectivity
ind = analyses.indices_connectivity_interbrain(subjall)

# cal. the connectivity matrix from the single participant
con = stats.con_matrix(subj0, freqs_mean)

# cal. the inter-brian metaconnectivity
meta = stats.metaconn_matrix_2brains(ind, con.ch_con, freqs_mean)

# permutation
dyad_all = [group1, group2]
sig = stats.statscluster(data = dyad_all, 
                   test = 'ind ttest', 
                   factor_level = None,
                   ch_con_freq = scipy.sparse.bsr_matrix(meta[0]),
                   tail = 0, 
                   n_permutations = 5000,
                   alpha = 0.05)
```
Although the above code is working, is it the right way to conduct the statistical analysis? It seems the connectivity matrix was not influenced by the frequencies (α/θ/β...) but only the numbers of bins in the frequency ROI.

Also, in the section of "Comparing Inter-brain connectivity values to random signal" of the getting_started, the argument was set ch_con_freq=None, what's the difference between ch_con_freq being set and remitted? 

Thanks a lot.

etherealsunshine, 2023-07-20:
Hi! I'll try my best to get back to you with the rest of your questions, but Setting ch_con_freq to None, as seen in the tutorial section, means that the function is not using a specific connectivity matrix for the permutation test. Instead, it compares the observed inter-brain connectivity values to a random signal.

deep-introspection, 2025-07-03:
Indeed, providing a `ch_con_freq` matrix or setting it to `None` leads to distinct behaviors:
- When `ch_con_freq=meta` is provided, the cluster-based permutation test leverages the specified adjacency structure across channels and frequency bins.
- When `ch_con_freq=None`, no adjacency is used; each feature is tested independently against a null distribution generated via permutation.

The former is therefore suitable for detecting structured effects i.e., clusters that are spatially or spectrally contiguous, while the latter yields a more conservative, global test without cluster-level inference.

Regarding your second question, the connectivity matrix does not reflect explicit frequency band labels because the input data is typically shaped as `(n_epochs, n_channels, n_freq_bins)`. Functions such as `compute_sync()` and `metaconn_matrix_2brains()` operate on the product of channel neighborhood combined with frequency bins neighborhood. This design choice ensures maximal flexibility across use cases, but it requires the user to manage band interpretation externally.