ppsp-team/HyPyP #165: Bugs in stats.statscluster

parkinsonlyn, 2023-05-15:
Hi HyPyP team, when using the stats.statscluster fuction, some problems occured:

1. The source code called the `mne.stats.permutation_cluster_test` to compute the cluster permutation t test, where `connectivity=ch_con_freq` is determined. However error showed no argument of "connectivity" defined. From [the function page of MNE](https://mne.tools/stable/generated/mne.stats.permutation_cluster_test.html), is the "adjacency" the right argument? 
2. When using "ind ttest" to calculate, the result suggested a significant cluster with mostly negative t-values included. But the values in this cluster didn't show in the "Stat_obs_plot" but only 0 and -0.

etherealsunshine, 2023-07-23:
Hi there, could you please describe the error message?

micmar15sr, 2023-07-24:
for the Issue n.1 I found the same error, when using  stats.statscluster you get the following error:

```python
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[305], line 6
      2 groupmat2,channels=ImportGroupFromXls(path2,freq,returnchannel=True)
      4 data=[groupmat1,groupmat2]
----> 6 Stat_obs, clusters, cluster_pv, H0, Stat_obs_plot=stats.statscluster(data=data,test='f oneway',factor_level=None,ch_con_freq=sparse,tail=0,n_permutations=1000,alpha=0.025)

File ~/anaconda3/envs/hypyp/lib/python3.11/site-packages/hypyp/stats.py:467, in statscluster(data, test, factor_level, ch_con_freq, tail, n_permutations, alpha)
    456 '''
    457 Stat_obs, clusters, cluster_p_values, H0 = mne.stats.permutation_cluster_test(data,
    458                                                                               stat_fun=stat_fun,
   (...)
    464                                                                               out_type='mask')
    465 '''
    466 #####NEW
--> 467 Stat_obs, clusters, cluster_p_values, H0 = mne.stats.permutation_cluster_test(data,
    468                                                                               stat_fun=stat_fun,
    469                                                                               threshold=alpha,
    470                                                                               tail=tail,
    471                                                                               n_permutations=n_permutations,
    472                                                                               connectivity=ch_con_freq,
    473                                                                               t_power=1,
    474                                                                               out_type='mask')
    476 # getting F values for sensors belonging to a significant cluster
    477 Stat_obs_plot = np.zeros(Stat_obs.shape)

TypeError: permutation_cluster_test() got an unexpected keyword argument 'connectivity'
```

however modifying the argument `connectivity` with `adjacency` in the source code the function accept the argument. Is it maybe due to an old version of MNE?
I would suggest the following change.
```python
 ###OLD
    '''
    Stat_obs, clusters, cluster_p_values, H0 = mne.stats.permutation_cluster_test(data,
                                                                                  stat_fun=stat_fun,
                                                                                  threshold=alpha,
                                                                                  tail=tail,
                                                                                  n_permutations=n_permutations,
                                                                                  connectivity=ch_con_freq,
                                                                                  t_power=1,
                                                                                  out_type='mask')
    '''
    #####NEW
    Stat_obs, clusters, cluster_p_values, H0 = mne.stats.permutation_cluster_test(data,
                                                                                  stat_fun=stat_fun,
                                                                                  threshold=alpha,
                                                                                  tail=tail,
                                                                                  n_permutations=n_permutations,
                                                                                  adjacency=ch_con_freq,
                                                                                  t_power=1,
                                                                                  out_type='mask')
```

deep-introspection, 2023-07-27:
@micmar15sr and @parkinsonlyn, it seems the error is caused by a mismatch of version. If you update HyPyP and MNE, it should work!

closed 2023-07-27 by deep-introspection

micmar15sr, 2023-07-28:
I have updated Hypyp but it seems that the error is not solved, I have 
### **hypyp=0.5.0b0** and **mne=1.4.2.** 
Are these the correct versions?

micmar15sr, 2023-07-28:
Maybe it has been fixed for **statsCondCluster** and not for **statsCluster** . 
Indeed inside  **statsCondCluster**  the mne function _permutation_cluster_test_ here reported, has the correct arguments. 
```python
    # computing the cluster permutation t test
    F_obs, clusters, cluster_p_values, H0 = permutation_cluster_test(data,
                                                                     threshold=None,
                                                                     n_permutations=n_permutations,
                                                                     tail=tail, adjacency=ch_con_freq,
                                                                     t_power=1, out_type='mask')

``` 
While **statsCluster** has not been correctly updated 

```python
 # computing the cluster permutation t test
    Stat_obs, clusters, cluster_p_values, H0 = mne.stats.permutation_cluster_test(data,
                                                                                  stat_fun=stat_fun,
                                                                                  threshold=alpha,
                                                                                  tail=tail,
                                                                                  n_permutations=n_permutations,
                                                                                  connectivity=ch_con_freq,
                                                                                  t_power=1,
                                                                                  out_type='mask')

```

deep-introspection, 2023-07-31:
I just upgraded the package. @micmar15sr can you please try now?

micmar15sr, 2023-07-31:
yes, thank you