ppsp-team/HyPyP #170: Bugs in Tutorial

etherealsunshine, 2023-05-30:
While running the tutorial I caught an AttributeError stating : 'NoneType' object has no attribute 'squeeze' while running the second code snippet.

```py
PDC = analyses.compute_conn_mvar(no_ICA_complex_signal, 
                                 mvar_params={"mvar_order": 2, "fitting_method":"default", "delta": 0},
                                 ica_params={"method": "infomax_extended", "random_state": None},
                                 measure_params={"name": "pdc", "n_fft": 512}
                                 )
```

```py
no_ICA_result_inter = []
for i in [0, 1]:
    PDC_q = PDC.squeeze()
    if i == 0 :
        mvar_spectrum =  PDC_q[n_ch:n_ch*2, 0:n_ch, :]
    else:
        mvar_spectrum =  PDC_q[0:n_ch, n_ch:n_ch*2, :]   
    PDC_fmax = np.zeros((n_ch, n_ch), dtype=PDC.dtype)
    for j in range(0, n_ch):
        for k in range(0, n_ch):
            PDC_fmax[j, k] = np.amax(mvar_spectrum[j,k,:])
    mvar_values_inter = PDC_fmax
    mvar_values_inter -= np.diag(np.diag(mvar_values_inter))
    # computing Cohens'D for further analyses for example
    mvar_C_inter = (mvar_values_inter -
               np.mean(mvar_values_inter[:])) / np.std(mvar_values_inter[:])
    # can also sample CSD values directly for statistical analyses
    no_ICA_result_inter.append(mvar_C_inter)
```

etherealsunshine, 2023-06-22:
This error will come up if you  do not answer the input when the code asks a question(and just press enter)

closed 2023-06-22 by etherealsunshine

GrimmieLynn, 2023-07-17:
Meet the same problem. And when I try to print PDC, it seems that the PDC is None type, but the no_ICA_complex_signal is fine. Did you solve the problems?

etherealsunshine, 2023-07-17:
Hey there,

Could you check if you answered the input prompt as shown below? If you answer it should work.
[image]

etherealsunshine, 2023-07-17:
Also if you're facing issues with ```numpy bool```, in the latest versions it's deprecated, so you'll have to downgrade

GrimmieLynn, 2023-07-19:
Hi! I tried to run the code, and this is what I answered:[image: image.png]
And it is running
[image: image.png]
, but afer that we still get the error message.[image: image.png]
And if you print(PDC), it will show none.
[image: image.png]
Also this is what I get after I tried to uninstall numpy 1.24.3 and install
numpy 1.24.1.
Sorry to interrupt you. Hope you have a nice day!

etherealsunshine, 2023-07-19:
Hi! I dont think the image was actually uploaded, could you share it again?