ppsp-team/HyPyP #172: Question related to ICA fit

imogenweigall, 2023-07-10:
Hi HyPyP team, thank you for your efforts in creating this great package. I have a question regarding your ICA_fit function. I am interested in why this function fits the ICA twice, first excluding bad channels and then a second time including bad channels. The typical recommendation is that bad channels are marked prior to ICA (https://mne.tools/stable/auto_tutorials/preprocessing/15_handling_bad_channels.html). If you could please provide me with further information / help me understand the rationale behind this it would be greatly appreciated.

```
icas = []
    for epoch in epochs:
        # per subj
        # applying AR to find global rejection threshold
        reject = get_rejection_threshold(epoch, ch_types='eeg')
        # if very long, can change decim value
        print('The rejection dictionary is %s' % reject)
        # fitting ICA on filt_raw after AR
        ica = ICA(n_components=n_components,
                  method=method,
                  fit_params= fit_params,
                  random_state=random_state).fit(epoch)
        # take bad channels into account in ICA fit
        epoch_all_ch = mne.Epochs.copy(epoch)
        epoch_all_ch.info['bads'] = []
        epoch_all_ch.drop_bad(reject=reject, flat=None)
        icas.append(ica.fit(epoch_all_ch)) 
```

etherealsunshine, 2023-07-17:
Hey there! 
(Disclaimer: Not a member of the initial team)

I think it's done to ensure robustness to the presence of bad channels, especially where bad channels may only become apparent after running ICA

deep-introspection, 2025-05-08:
Thank you for your question @imogenweigall!

The ICA_fit function isn't actually fitting ICA twice - rather, it's using a specific approach to handle bad channels during the fitting and application steps:
1. When fitting the ICA model, it temporarily excludes bad channels by creating a copy of the epochs and setting epochs_fit.info['bads'] = []. This is because including bad channels in the ICA decomposition could negatively influence the quality of the decomposition.
2. When applying the ICA back to the data (which happens in the ICA_apply function), it restores the original bad channels information with epochs_subj.info['bads'] = bads_keep.

This approach differs slightly from MNE's standard recommendation to mark bad channels prior to ICA for a specific reason. HyPyP is designed for hyperscanning (simultaneous recording from multiple participants), and maintaining a consistent data structure between participants is critical for later inter-brain analyses.

closed 2025-05-08 by deep-introspection