ppsp-team/HyPyP #174: getting_started.ipynb Attribute Error

JosieKearney, 2023-07-31:
Hi HyPyP team!

Thank you for all your efforts in putting together this package.

I've been running the tutorial (using Python 3.11) to get to grips with the package and when I get to computing frequency- and time-frequency-domain connectivity measures obtained by MVARICA:

```
PDC = analyses.compute_conn_mvar(no_ICA_complex_signal, 
                                 mvar_params={"mvar_order": 2, "fitting_method":"default", "delta": 0},
                                 ica_params={"method": "infomax_extended", "random_state": None},
                                 measure_params={"name": "pdc", "n_fft": 512}
                                 )

no_ICA_result_inter = []
for i in [0, 1]:
    PDC_q = PDC.squeeze()
    if i == 0 :
        mvar_spectrum =  PDC_q[n_ch:n_ch*2, 0:n_ch, :]
    else:
        mvar_spectrum =  PDC_q[0:n_ch, n_ch:n_ch*2, :]   
    PDC_fmax = np.zeros((n_ch, n_ch), dtype=PDC.dtype)
    for j in range(0, n_ch):
        for k in range(0, n_ch):
            PDC_fmax[j, k] = np.amax(mvar_spectrum[j,k,:])
    mvar_values_inter = PDC_fmax
    mvar_values_inter -= np.diag(np.diag(mvar_values_inter))
    # computing Cohens'D for further analyses for example
    mvar_C_inter = (mvar_values_inter -
               np.mean(mvar_values_inter[:])) / np.std(mvar_values_inter[:])
    # can also sample CSD values directly for statistical analyses
    no_ICA_result_inter.append(mvar_C_inter)
    
    no_ICA_result_intra = []
```

 I keep get this error: 

`AttributeError: 'NoneType' object has no attribute 'squeeze'`

This is my first real attempt at using python for analysis, but my understanding is that this is happening because one of the functions above is returning 'None' and so  `PDC_q = PDC.squeeze()` cannot run.

 I inititally thought it might be 'ica_params' but changing this didn't fix the error, so I'm not sure why any of these functions are returning None in a way that would cause an error like this.

So my question is, what is causing this error and how do I go about resolving it? 

Thank you advance!
Josie

JosieKearney, 2023-08-09:
> 
Thanks for getting back to me

I've had a look at the advice given in #170 - I've answered the prompts (I get one for the ICA selection and one about merging epochs at the point of print PDC). I think it might be a NumPy issue, I have version 1.25.2 installed - should it be older/newer? Thanks in advance

etherealsunshine, 2023-08-10:
could you try numpy==1.23.1 and let me know if you still face any issues?

JosieKearney, 2023-08-11:
> 

Thank you again for your response - I have tried the older version of numpy, as well as reinstalling HyPyP and the error persists.

ghazalehran, 2024-08-19:
@JosieKearney Hi! :) I was wondering if you have enough samples. When running the code, it should tell you the minimum number of samples you would need for computing PDC, did you check this as well?

closed 2025-04-17 by deep-introspection