ppsp-team/HyPyP #182: utils.merge() does not save filter information

obergmartin, 2023-11-02:
When making a merged epoch object I would expect the high/low pass filter information to be the same. Or do I not understand the intended behaviour?

Here's some code that reproduces the result I'm finding:
```
data_path = Path(mne.datasets.sample.data_path(verbose=False))
sample_dir = data_path / "MEG" / "sample"
subjects_dir = data_path / "subjects"
raw_path = sample_dir / "sample_audvis_filt-0-40_raw.fif"
raw = mne.io.read_raw(raw_path).crop(10)
e = mne.make_fixed_length_epochs(raw, duration=1, preload=True)
# use the same data just for the example
e_merge = utils.merge(e,e)
print(e_merge.info['highpass'], e.info['highpass'])
print(e_merge.info['lowpass'], e.info['lowpass'])
# the output:
# (0.0, 0.10000000149011612)
# (75.07687377929688, 40.0)
```

I think the that can be fixed here:?
https://github.com/ppsp-team/HyPyP/blob/8ecb507b40378aa839fa90d4a0b874a48b8e4162/hypyp/utils.py#L173

obergmartin, 2023-11-03:
Is it safe to do this?
```
with info._unlock():
    info["lowpass"] = the_original_value
```

deep-introspection, 2025-05-25:
Ok, I updated the dev version of HyPyP.
It will be put in the next PyPI version.
Thanks @obergmartin!

You can test with:

```python
import mne
mne.utils.use_log_level('error')
from pathlib import Path
from hypyp import utils

import warnings
warnings.filterwarnings("ignore", category=RuntimeWarning)

# Your test code from the issue
data_path = Path(mne.datasets.sample.data_path(verbose=False))
sample_dir = data_path / "MEG" / "sample"
raw_path = sample_dir / "sample_audvis_filt-0-40_raw.fif"
raw = mne.io.read_raw(raw_path).crop(10)
e = mne.make_fixed_length_epochs(raw, duration=1, preload=True)

# Use the fixed merge function
e_merge = utils.merge(e, e)

print("Original epochs:", e.info['highpass'], e.info['lowpass'])
print("Merged epochs:", e_merge.info['highpass'], e_merge.info['lowpass'])

# Should now show identical values
assert e_merge.info['highpass'] == e.info['highpass']
assert e_merge.info['lowpass'] == e.info['lowpass']
```

closed 2025-06-05 by deep-introspection