ppsp-team/HyPyP #203: Save topomaps as svg image

thiago-roque07, 2025-02-21:
Hi, 
I recently added this feature to my own fork of HyPyP to facilitate the use of these plots for publication.
I wonder if other people would be interested in this feature to start a pull request to here

Ramdam17, 2025-04-04:
Hello Thiago,

Thank you for reaching out !

My name is Rémy, I've recently joined the project along with Patrice as engineers responsible for maintaining and further developing HyPyP.

Currently, we're ensuring the repository remains up-to-date, and plan to release an upgraded version in the next 6 months (introducing object-oriented structure, pattern injection, parallel processing, etc.).

The SVG export feature for topomaps you've implemented would be really useful and we would love to integrate it in the next release.

Would you be open to collaborating with us in the coming months to integrate this feature?

Best regards,
Rémy

thiago-roque07, 2025-04-08:
Hi Rémy,

It will be a pleasure to collaborate with the project!
I've been using it extensively, and have been working on some personal
customizations that might be interesting to others. Including some
integration between HyPyP and MNE-LSL for real time assessment of
inter-brain coupling (which is basically the core of my PhD thesis).
But I'm on the verge of my Comprehensive Exam, for my PhD, so I'll only be
available to collaborate after the end of this semester.

Best regards,
--
Thiago R. Roque

Ramdam17, 2025-05-14:
Hi Thiago,

Thank you for your reply, and best of luck for your Comprehensive Exam!

We’ll definitely reach out as soon as we enter the development cycle for the next release. It would be a real pleasure to collaborate if you find the time.

Best regards,
Rémy

closed 2025-05-14 by Ramdam17